flipperdevices/blackmagic-esp32-s2 #13: Missing driver folder for building firmware

0wlyW00d, 2022-07-25:
Hello,

following your tutorial to flash the dev-board to the stock firmware , I got an error when running `idk.py build`

`Executing action: all (aliases: build)
Running ninja in directory /Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2/build
Executing "ninja all"...
[1/174] Building C object esp-idf/blackmagic/CMakeFiles/__idf_blackmagic.dir/esp32-platform/platform.c.objFAILED: esp-idf/blackmagic/CMakeFiles/__idf_blackmagic.dir/esp32-platform/platform.c.obj
/Users/guillaume/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc  -I/Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2/build/config -I/Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2/components/blackmagic/blackmagic-fw/src/include -I/Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2/components/blackmagic/blackmagic-fw/src/platforms/common -I/Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2/components/blackmagic/blackmagic-fw/src/target -I/Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2/components/blackmagic/blackmagic-fw/libopencm3/include -I/Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2/components/blackmagic/esp32-platform -I/Users/guillaume/esp/esp-idf/components/newlib/platform_include -I/Users/guillaume/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/guillaume/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/Users/guillaume/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/guillaume/esp/esp-idf/components/freertos/esp_additions/include -I/Users/guillaume/esp/esp-idf/components/esp_hw_support/include -I/Users/guillaume/esp/esp-idf/components/esp_hw_support/include/soc -I/Users/guillaume/esp/esp-idf/components/esp_hw_support/include/soc/esp32s2 -I/Users/guillaume/esp/esp-idf/components/esp_hw_support/port/esp32s2/. -I/Users/guillaume/esp/esp-idf/components/esp_hw_support/port/esp32s2/private_include -I/Users/guillaume/esp/esp-idf/components/heap/include -I/Users/guillaume/esp/esp-idf/components/log/include -I/Users/guillaume/esp/esp-idf/components/soc/include -I/Users/guillaume/esp/esp-idf/components/soc/esp32s2/. -I/Users/guillaume/esp/esp-idf/components/soc/esp32s2/include -I/Users/guillaume/esp/esp-idf/components/hal/esp32s2/include -I/Users/guillaume/esp/esp-idf/components/hal/include -I/Users/guillaume/esp/esp-idf/components/hal/platform_port/include -I/Users/guillaume/esp/esp-idf/components/esp_rom/include -I/Users/guillaume/esp/esp-idf/components/esp_rom/include/esp32s2 -I/Users/guillaume/esp/esp-idf/components/esp_rom/esp32s2 -I/Users/guillaume/esp/esp-idf/components/esp_common/include -I/Users/guillaume/esp/esp-idf/components/esp_system/include -I/Users/guillaume/esp/esp-idf/components/esp_system/port/soc -I/Users/guillaume/esp/esp-idf/components/esp_system/port/include/private -I/Users/guillaume/esp/esp-idf/components/xtensa/include -I/Users/guillaume/esp/esp-idf/components/xtensa/esp32s2/include -I/Users/guillaume/esp/esp-idf/components/lwip/include -I/Users/guillaume/esp/esp-idf/components/lwip/include/apps -I/Users/guillaume/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/guillaume/esp/esp-idf/components/lwip/lwip/src/include -I/Users/guillaume/esp/esp-idf/components/lwip/port/esp32/include -I/Users/guillaume/esp/esp-idf/components/lwip/port/esp32/include/arch -mlongcalls  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-error=format= -Wno-format -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=/Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2=. -fmacro-prefix-map=/Users/guillaume/esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0-dev-4379-g36f49f361c\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -DPC_HOSTED=0 -DFIRMWARE_VERSION=\"v1.7.1-284-g7307f08\" -Wno-char-subscripts -Wno-attributes -MD -MT esp-idf/blackmagic/CMakeFiles/__idf_blackmagic.dir/esp32-platform/platform.c.obj -MF esp-idf/blackmagic/CMakeFiles/__idf_blackmagic.dir/esp32-platform/platform.c.obj.d -o esp-idf/blackmagic/CMakeFiles/__idf_blackmagic.dir/esp32-platform/platform.c.obj -c /Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2/components/blackmagic/esp32-platform/platform.c
/Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2/components/blackmagic/esp32-platform/platform.c:6:10: fatal error: driver/gpio.h: No such file or directory
    6 | #include <driver/gpio.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
[1/1] cd /Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2/build/bootloader/esp-idf/esptool...er 0x1000 /Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2/build/bootloader/bootloader.binBootloader binary size 0x35e0 bytes. 0x3a20 bytes (52%) free.
[12/172] Building C object esp-idf/blackmagic/CMakeFiles/__idf_blackmagic.dir/blackmagic-fw/src/target/stm32f1.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2/build/log/idf_py_stderr_output_27664 and /Users/guillaume/Dev/flipperzero/blackmagic-esp32-s2/build/log/idf_py_stdout_output_27664`

It appears that the folder "driver" and the file "gpio.h" is missing in the GitHub repository, even cloning it with the '--recursive'

skotopes, 2022-07-25:
Did you clone it recursively or initialized git submodules?

0wlyW00d, 2022-07-25:
as in the README.md file on a clean folder I used the `git clone --recursive https://github.com/flipperdevices/blackmagic-esp32-s2.git` commands

skotopes, 2022-07-25:
Can you how `git submodule status`

0wlyW00d, 2022-07-25:
yeah sure, here's the output:

`➜  blackmagic-esp32-s2 git:(dev) ✗ git submodule status
 7307f086c6425ee8a91fa2c3cd69c210a3cd165d components/blackmagic/blackmagic-fw (v1.6.1-866-g7307f08)
 9549e82b048e93167529319ca1d169f3db1d2572 components/mlib/mlib (V0.5.0-222-g9549e82)
 dd30f2c648b984949b858d7dcc914297985a20bb components/tinyusb/tinyusb (0.12.0-188-gdd30f2c64)`

skotopes, 2022-07-25:
Which version of esp idf do you use?

0wlyW00d, 2022-07-25:
Actually using the last idf version 
`➜  blackmagic-esp32-s2 git:(dev) ✗ idf.py --version
ESP-IDF v5.0-dev-4379-g36f49f361c`

installed as requested on the Espressif website : https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html

skotopes, 2022-07-25:
Please switch to stable branch v4.4

closed 2022-07-25 by skotopes

skotopes, 2022-07-25:
And reopen if it will not help. Meanwhile I'll update readme.