flipperdevices/blackmagic-esp32-s2 #21: Unable to flash: driver/gpio.h: not found

alvarotorijano, 2023-01-29:
On a fresh w10 install there is no way to properly run idf.py -p COM14 flash because it complains about this error:
blackmagic-esp32-s2/components/blackmagic/esp32-platform/platform.c:6:10: fatal error: driver/gpio.h: No such file or directory

Why is not esp-idf being able to find this header?
a W10 guide on how to flash the debugger firmware would be really appreciated

alvarotorijano, 2023-02-01:
just check this out https://github.com/alvarotorijano/FlipperBlackMagicReady

DrZlo13, 2023-02-07:
It seems you are trying to compile with idf 5.0. You need esp-idf 4.4 to build the project.

alvarotorijano, 2023-02-07:
Installing the enviroment is at least a pain in the ass. If you don't select the proper microcontroller selection you have to manually remove it to be able to reinstall again. The install script also needs pip3 and python 3.8. There are broken dependencies that are not installed. The install script is also not able to clone properly all the subrepositories, some of them was needed to be cloned y hand. In this repository there are no notes about the required esp-idf version to be used. 
After crying for 3 days I compiled it and uploaded the binaries which could worth to upload here as a release. 
Hope this helps any wandering soul. 
https://github.com/alvarotorijano/FlipperBlackMagicReady

DrZlo13, 2023-02-08:
Okay, I'll update the build documentation.

DrZlo13, 2023-02-09:
Fixed in #22

closed 2023-02-09 by DrZlo13