flipperdevices/blackmagic-esp32-s2 #24: mdns does not work in AP mode

11wanqin, 2023-02-18:
The mdns works fine in STA mode, but it seems does not work in AP mode.
if move "network_post_init();" in front of  “switch(wifi_mode)”，the mdns can work fine in AP and STA mode.
[image]
[image]
[image]

DrZlo13, 2023-03-20:
Thanks for the report, we'll fix it as soon as we have time.

11wanqin, 2023-03-20:
Thank you for getting back to me. This is an excellent open-source project; I hope it can improve.

DrZlo13, 2023-03-20:
I am reopening this issue to track it down.

closed 2023-03-21 by DrZlo13

DrZlo13, 2023-03-21:
Fixed in #29