flipperdevices/blackmagic-esp32-s2 #26: Read logs?

lo00l, 2023-03-20:
Hi there,
What tools should I use to read logs?

I can see the logging is performed via uart 7th pin. So I use my USB-to-TTL converter connecting IO7 pin to TX pin of my converter. The I run `screen /dev/cu.usbserial-2130 57600`.
Everything works except log messaged are printed with really strange formatting (see screenshot). It makes it hardy readable.
Can I fix it somehow? Thank you.
[image]

lomalkin, 2023-03-20:
Hello, you should reset terminal first via command `reset`, and try again with screen.

DrZlo13, 2023-03-20:
Also, UART baud rate is 230400

DrZlo13, 2023-03-20:
I use miniterm that comes with pyserial, like `python3 -m serial.tools.miniterm --raw  /dev/cu.usbmodem112203 57600`

lo00l, 2023-03-20:
@lomalkin running `reset` before `screen` didn't help  :(

@DrZlo13 yeah, with miniterm everything works great, thank you 👍

closed 2023-03-20 by lo00l