flipperdevices/blackmagic-esp32-s2 #41: Instable DAPLink on ESP32-S2 evalboard?

rena2019, 2023-06-02:
Seems that the DAPLink interface is not stable on my Muselab  NanoESP32 S2 evalboard even if I set the adapter speed down to 1kHz:
```
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x303a:0x4002, serial=DAP_7CDFA105F2D6
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Serial# = 7CDFA105F2D6
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 1 kHz
Info : SWD DPIDR 0x2ba01477
Info : [STM32F401CEUx.cpu] Cortex-M4 r0p1 processor detected
Info : SWD DPIDR 0x2ba01477
Error: Failed to read memory at 0xe000ee00
Info : SWD DPIDR 0x2ba01477
Error: Failed to read memory and, additionally, failed to find out where
Info : [STM32F401CEUx.cpu] target has 0 breakpoints, 0 watchpoints
Info : SWD DPIDR 0x2ba01477
Info : SWD DPIDR 0x2ba01477
Error: Failed to read memory and, additionally, failed to find out where
Error: read_memory: read at 0xe0042004 with width=32 and count=1 failed
Error executing event examine-end on target STM32F401CEUx.cpu:
Info : starting gdb server for STM32F401CEUx.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : SWD DPIDR 0x2ba01477
Polling target STM32F401CEUx.cpu failed, trying to reexamine
Info : SWD DPIDR 0x2ba01477
Info : [STM32F401CEUx.cpu] Cortex-M4 r0p1 processor detected
Info : SWD DPIDR 0x2ba01477
Error: Failed to read memory and, additionally, failed to find out where
Examination failed, GDB will be halted. Polling again in 100ms
```

skotopes, 2023-09-26:
Try to check cables first. If you have oscilloscope check data exchange.

KondratokGK, 2024-01-21:
Same issue. It`s not working with stm32cubeide via wifi or usb. But Flipper Zero as Daplink working perfectly/

Hollywood-bit, 2024-09-19:
I’m sorry don’t no we’re or what to do I bout cards whit
Esp32/gps/nrf24/cciidiver3 don’t no how to install or anything very new have no clue please need lots help

skotopes, 2024-09-19:
> I’m sorry don’t no we’re or what to do I bout cards whit Esp32/gps/nrf24/cciidiver3 don’t no how to install or anything very new have no clue please need lots help

Start with official documentation: https://docs.flipper.net

aravind-hash, 2026-10-28:
Close