flipperdevices/blackmagic-esp32-s2 #43: Unable to attach to blackmagic using flipper build tool.

cnburke, 2023-07-10:
Hi. I'm trying to work my way through the flipper wifi dev board setup instructions, but I'm unable to attach to the running blackmagic process via USB or WiFi. My dev board seems to be working only if I manually compile blackmagic myself and use the idf.py tool to build and monitor. I have tried every combination of serial port and boot mode the instructions offered, as well as trying the WiFi direct and direct USB methods. If anyone could let me know if I'm doing something stupid here, I'd greatly appreciate it. 

The log at the end is taken from flipper's VS Code Debug configuration with the "showDevDebugOutput": "raw" option uncommented. 

I'm running `./fbt flash_blackmagic BLACKMAGIC=/dev/cu.usbmodem01` before attempting to debug. 

I am able to achieve monitoring via 

```
➜  blackmagic-esp32-s2 git:(dev) ✗ idf.py -p /dev/cu.usbmodem01 monitor        
Executing action: monitor
Running idf_monitor in directory /Users/cameron/code-projects/flipper/blackmagic-esp32-s2
Executing "/Users/cameron/.espressif/python_env/idf4.4_py3.11_env/bin/python /Users/cameron/code-projects/flipper/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbmodem01 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 /Users/cameron/code-projects/flipper/blackmagic-esp32-s2/build/blackmagic.elf -m '/Users/cameron/.espressif/python_env/idf4.4_py3.11_env/bin/python' '/Users/cameron/code-projects/flipper/esp/esp-idf/tools/idf.py' '-p' '/dev/cu.usbmodem01'"...
--- idf_monitor on /dev/cu.usbmodem01 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
```

If it helps, in my attempts, I have seen the error of "'monitor' command not supported on this target" (or something along those lines, pulling from memory as I can't get the device back into whatever state this was). 

Log output trying to debug the running process:
```
Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from /Users/cameron/code-projects/flipper/flipperzero-firmware/toolchain/x86_64-darwin/bin/arm-none-eabi-objdump --syms -C -h -w /Users/cameron/code-projects/flipper/flipperzero-firmware/build/latest/firmware.elf
Reading symbols from /Users/cameron/code-projects/flipper/flipperzero-firmware/toolchain/x86_64-darwin/bin/arm-none-eabi-nm --defined-only -S -l -C -p /Users/cameron/code-projects/flipper/flipperzero-firmware/build/latest/firmware.elf
Launching GDB: /Users/cameron/code-projects/flipper/flipperzero-firmware/toolchain/x86_64-darwin/bin/arm-none-eabi-gdb-py -q --interpreter=mi2
1-gdb-version
Finished reading symbols from objdump: Time: 63 ms
Finished reading symbols from nm: Time: 272 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2023.06) 10.2.90.20210621-git\n"
-> ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-apple-darwin10 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source /Users/cameron/.vscode/extensions/marus25.cortex-debug-1.12.0/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source /Users/cameron/.vscode/extensions/marus25.cortex-debug-1.12.0/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-interpreter-exec console "set output-radix 0xa"
-> ~"Output radix now set to decimal 10, hex a, octal 12.\n"
Output radix now set to decimal 10, hex a, octal 12.
-> 8^done
9-interpreter-exec console "set input-radix 0xa"
-> ~"Input radix now set to decimal 10, hex a, octal 12.\n"
Input radix now set to decimal 10, hex a, octal 12.
-> 9^done
10-file-exec-and-symbols "/Users/cameron/code-projects/flipper/flipperzero-firmware/build/latest/firmware.elf"
-> 10^done
11-target-select extended-remote /dev/cu.usbmodem101
-> 11^connected
12-interpreter-exec console "monitor halt"
-> @"Cortex-M timeout to wait for device halts: 2000\n"
Cortex-M timeout to wait for device halts: 2000
-> 12^done
13-interpreter-exec console "monitor swdp_scan"
-> @"SW-DP scan failed!\n"
SW-DP scan failed!
-> @"Failed\n"
Failed
-> 13^done
14-interpreter-exec console "attach 1"
-> ~"Attaching to program: /Users/cameron/code-projects/flipper/flipperzero-firmware/build/f7-firmware-D/firmware.elf, Remote target\n"
Attaching to program: /Users/cameron/code-projects/flipper/flipperzero-firmware/build/f7-firmware-D/firmware.elf, Remote target
-> &"Attaching to Remote target failed\n"
Attaching to Remote target failed
-> 14^error,msg="Attaching to Remote target failed"
15-interpreter-exec console "set confirm off"
Failed to launch GDB: Attaching to Remote target failed (from interpreter-exec console "attach 1")
```

SuperJesseH, 2023-08-11:
Having the same issues... will post if I find a solution

alvarotorijano-safeloc, 2023-08-18:
Same problem here, any idea?

alvarotorijano-safeloc, 2023-08-21:
Ok, after checking with other boards seems to be an electrical problem. In this case the wifi dev board is not properly connected to the flipper. This can me due to a bad position or a damaged internal connector.
Dissasembly your flipper and check with the polymeter.
Please let me know if this is useful.

Best regards!

cnburke, 2023-08-21:
The Settings->System->Debug ON worked for me

closed 2023-08-21 by cnburke